stinos/micropython-wrap #1: Delta with current micropython version ?

touilleMan, 2016-12-08:
Trying to use this lib, it seems it is only compatible with an old version of micropython where `mp_obj_fun_builtin_fixed_t` and `mp_obj_fun_builtin_fixed_t` where still part of `py/obj.h`

So I was wondering which version of micropython works with this lib and how big was the delta to make it work against a recent micropython.

stinos, 2016-12-08:
Not sure what you mean: commit https://github.com/stinos/micropython-wrap/commit/44bb9148130ce543ec621f4a6d85554d8058f635 was there to adapt to uPy after https://github.com/micropython/micropython/pull/2529 (no `mp_obj_fun_builtin_t` anymore but instead `mp_obj_fun_builtin_fixed_t` and `mp_obj_fun_builtin_var_t`) so this should compile against the most recent uPy version. It's been a while I tested everything on unix though, is that a problem maybe? How are you building this and on what platform?

touilleMan, 2016-12-09:
My bad ! I didn't realize my version of micropython was outdated. With micropython v1.6.8, you library works like a charm ^^

closed 2016-12-09 by touilleMan

stinos, 2016-12-09:
Well, sort of, I checked the gcc build again yesterday and it doesn't compile currently but I'll fix that shortly.