stinos/micropython-wrap #3: Wrapping the TinyGSM module on ESP32

amotl, 2020-02-14:
Dear @stinos,

thank you very much for conceiving and maintaining this fine piece of software.

I am currently thinking about whether it might be possible to use it in order to bring the excellent [TinyGSM](https://github.com/vshymanskyy/TinyGSM) library to the world of MicroPython.

Within [1], you are saying that it has been tested on Unix and Windows, so I am wondering if there would be a chance to also use it on the ESP32 port for MicroPython.

I am aware there surely will be some intrinsic things to care about regarding runtime interaction with FreeRTOS and such aka. "the devil is in the details" when actually starting this endeavor. I just wanted to bring up the idea in general and to get some feedback whether you believe the undertaking could be feasible in general.

Thanks already for taking the time!

With kind regards,
Andreas.

[1] https://forum.micropython.org/viewtopic.php?t=791#p4592

cc @vshymanskyy, @SRGDamia1

SRGDamia1, 2020-02-14:
I'm sorry, I haven't done any work at all with MicroPython (or with an ESP32) and don't really have time to work on it right now.  If it's as simple as creating an extra header file or adding a struct at the bottom that list out all the functions for each module, I can do that, but I can't do much more.

I know @vshymanskyy has done a lot more with the ESP32, but I don't know if he's done any of it in MicroPython.

Just FYI I'm finishing up some restructuring of TInyGSM to use CRTP to reduce duplication between modules, so if porting is as simple as tacking structs to the headers, now is definitely the time to ask me to do it.

stinos, 2020-02-15:
> Within [1], you are saying that it has been tested on Unix and Windows, so I am wondering if there would be a chance to also use it on the ESP32 port for MicroPython.

Any compiler which is roughly C++11 compatible should work. ESP32 uses gcc as far as I know so that should be covered. There are no special runtime features, it's just C++ code depending on the standard libraries, so whether this runs on an microcontroller or PC or FreeRTOS doesn't matter.

But, this library is only about wrapping C or C++ functions and classes. To get something like TinyGSM available as a module in MicroPython you also need to get that wrapper code built into a module which can be loaded into the MicroPython process. This library was written before there was any support for external module features in MicroPython, so back then I added that using standard dynamic library loading as found on unix and windows but that is not something which works on ESP32, I think.

Since then MicroPython has grown functionality to support external C modules though in 2 forms:
- there's http://docs.micropython.org/en/latest/develop/cmodules.html which should work for this usecase but requires the module being built into MicroPython
- recently MicroPython is getting support for native dynamic modules http://docs.micropython.org/en/latest/develop/natmod.html, which is nice because it builds a separate .mpy file without so it's sort of ideal for usecases like this. However such native code requires the API to be accessed through py/dynruntime.h instead of py/runtime.h. The latter is what micropython-wrap currently uses. It should only be a relatively small change to make it use the former and I'd be happy to change that, but the problem is dynruntime.h does not yet implement the complete API and as such micropython-wrap won't compile with it.

amotl, 2020-02-18:
Dear @SRGDamia1 and @stinos,

thanks for your valuable answers around that topic. I've just created https://github.com/vshymanskyy/TinyGSM/issues/362 in order to signal interest about this to the users and contributors of TinyGSM.

I hope you don't mind keeping these notes around, otherwise feel free to close them right away. However, I see there might be a chance others are researching about the same topic these days, who could then continue here.

Thanks again and with kind regards,
Andreas.

stinos, 2020-02-20:
@amotl I'm going to close this for now (it remains on github so still discoverable), because the basic answer is 'yes this should work'. Feel free to reopen in case you have any actual problems.

closed 2020-02-20 by stinos

amotl, 2020-02-20:
Sure. Thanks a bunch about your feedback already!