stinos/micropython-wrap #6: ESP32

IhorNehrutsa, 2020-09-08:
Hi everyone.
I noticed issue #5. I work with micropython under Ubuntu. Some installation commands in [ubuntu_install.zip](https://github.com/stinos/micropython-wrap/files/5190844/ubuntu_install.zip)

I can compile examples from esp-idf and mpy-cross from micropython.
I try to compile micropython ESP32 port with modules.
[modules.zip](https://github.com/stinos/micropython-wrap/files/5190615/modules.zip)
The directory structure is:
```
$(HOME)/
   | esp-idf
   | micropython
   | micropython-wrap
   | modules
```

I added 
```
INC += -I$(HOME)
```
to the micropython/ports/esp32/Makefile.

The micropython-wrap is switched to ESP32-C++ branch.
The micropython is switched to    git hash (v3.3): 9e70825d1e1cbf7988cf36981774300066580ea7.

The 'example' module and 'temperature_conversion' module are compiled well and work after flashing 'firmware.bin' to the ESP32 generic board.
```
(build-venv) pc@ubuntu:~/micropython/ports/esp32$ make USER_C_MODULES=../../../modules all
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Including User C Module from ../../../modules/temperature_conversion
Including User C Module from ../../../modules/example
Building with ESP IDF v3
GEN build-GENERIC/genhdr/qstrdefs.collected.h
QSTR not updated
GEN build-GENERIC/genhdr/compressed.collected
Compressed data not updated
AR build-GENERIC/esp-idf/esp32/libesp32.a
LINK build-GENERIC/application.elf
   text	   data	    bss	    dec	    hex	filename
1099596	 278868	  37380	1415844	 159aa4	build-GENERIC/application.elf
Create build-GENERIC/application.bin
esptool.py v2.8
Create build-GENERIC/firmware.bin
bootloader     21344
partitions      3072
application  1378592
total        1444128
```

Compiling with 'foo' module gets errors:
```
(build-venv) pc@ubuntu:~/micropython/ports/esp32$ make USER_C_MODULES=../../../modules all
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Including User C Module from ../../../modules/foo
Including User C Module from ../../../modules/temperature_conversion
Including User C Module from ../../../modules/example
Building with ESP IDF v3
GEN build-GENERIC/genhdr/qstrdefs.collected.h
QSTR not updated
GEN build-GENERIC/genhdr/compressed.collected
Compressed data not updated
CC ../../../modules/foo/foo.c
In file included from /home/pc/micropython-wrap/detail/frompyobj.h:4:0,
                 from /home/pc/micropython-wrap/detail/callreturn.h:4,
                 from /home/pc/micropython-wrap/functionwrapper.h:4,
                 from ../../../modules/foo/foo.c:1:
/home/pc/micropython-wrap/detail/micropython.h:7:8: error: expected identifier or '(' before string constant
 extern "C"
        ^
In file included from /home/pc/micropython-wrap/detail/frompyobj.h:4:0,
                 from /home/pc/micropython-wrap/detail/callreturn.h:4,
                 from /home/pc/micropython-wrap/functionwrapper.h:4,
                 from ../../../modules/foo/foo.c:1:
/home/pc/micropython-wrap/detail/micropython.h:24:17: fatal error: cmath: No such file or directory
 #include <cmath>
                 ^
compilation terminated.
make: *** [../../py/mkrules.mk:63: build-GENERIC/foo/foo.o] Error 1
```
I see cmath files.
[image]
They notated with: 
// -*- C++ -*- C forwarding header.  

Where can I add the include path?
Thanks.

stinos, 2020-09-09:
> The micropython-wrap is switched to ESP32-C++ branch.
> The micropython is switched to git hash (v3.3): 9e70825d1e1cbf7988cf36981774300066580ea7.

That last one is not going to work, should also be my ESP-32-C++ branch, per the instructions I left in at https://github.com/stinos/micropython-wrap/issues/5#issuecomment-680870982

> rom ../../../modules/foo/foo.c:1:
> /home/pc/micropython-wrap/detail/micropython.h:7:8: error: expected identifier or '(' before string constant extern "C"

micropython-wrap is C++ code and must be compiled with a C++ compiler, you are compiling it as C code, which is also why it does not find cmath.h btw. Pleas try with the instructions mentioned above, and use C++ code in a file like foo.cpp, not foo.c.

stinos, 2020-10-06:
Closing this since another user could use the instructions in aforementioned issue to build and run the C++ module.

closed 2020-10-06 by stinos