stinos/micropython-wrap #9: wrapping c++ library

hstarmans, 2022-02-17:
I ran "make usercmodule MICROPYTHON_PORT_DIR=../micropython/ports/esp32" 
This creates a new binary, i flashed the binary to the chip, but how do i use the module?
I tried to import mod, foo etc. but it does not work?

stinos, 2022-02-18:
`import upywraptest` is the one, the Makefile is for building the tests.
I can't tell why that doesn't work for you, can you provide more information? MicroPython version used, does the compiler log say 'including user C module from ...', can you see it compiles tests/module.c and tests/module.cpp?

stinos, 2022-02-20:
>  Running the micropython-wrap makefile results in a directory build-usercmod in micropython/ports/esp32.

Can you doublecheck it effectively compiles both the .c and .cpp file in the tests/ directory? Either from the log of by introducing a compiler error in the files.
I'd also need to know the exact version of the ESP SDK you're using (I can only try to reproduce this if I have the same version of everything).

Btw do the micropython example modules work for you? Something like `make USER_C_MODULES=../../examples/usercmodule`

>  With the latest master your makefile runs into the following error;
> CMake Error at /home/hexastorm/projects/micropython-wrap/Makefile:11

That's actually MicroPython running into an error, it seems to be trying to use CMake to build a (gnu make) Makefile, that's never going to work. I'll have a look how to make that work with the most recent version of everything.

> did you test the code on a ESP32?

At one point yes, see see https://github.com/stinos/micropython-wrap/issues/5#issuecomment-704333711

> it seems to me I have to wrap all C++ code in extern c blocks which is a non-starter.

That's not the case, only the code which needs to go into the C module directly needs to be wrapped in `extern "c"`. See tests/module.cpp for instance. The micropython example is just done like that for simplicity. The function body could have been a separete 'pure' C++ function called from the C function.

hstarmans, 2022-02-21:
Stinos, my limited understanding of the interaction between c and  c++ seems indeed to be the issue.  I will work further with the example from micropython master.

closed 2022-02-21 by hstarmans

stinos, 2022-02-22:
Let me knoe if you'd get back to this and still have problems.

hstarmans, 2022-02-22:
@stinos , I will I made a Micropython wrapper for TMCStepper, see https://github.com/hstarmans/micropythonwebserver .
It compiles and most likely works. Still need to test it on actual hardware.  Thanks!