stinos/micropython-wrap #11: Support for keyword arguments in functions

JustinOng, 2022-09-13:
Hello,

Does this wrapper support keyword arguments in functions as described [here](https://micropython-usermod.readthedocs.io/en/latest/usermods_08.html#keyword-arguments-with-numerical-values)?

The closest I've found is optional variables, but that dosen't provide the same level of verbosity when passing arguments.

Thanks!

stinos, 2022-09-13:
That is not supported, main reason being that I don't know a way of generating the wrapper automatically, i.e. if a C function is `int func(int a, int b)` I wouldn't know how to prgrammatically get `a` and `b` strings in the call to the registration (`Def<funcName>(func)`) to generate the MicroPython boilerplate code you show.

It could probably be made to work if the registration would take the keyword names (something like `Def<funcName>(func, "a", "b")`) but I haven't looked into that. Though I admit I've wanted it a couple of times..

stinos, 2022-09-22:
@JustinOng had a look at this and it turned out to be relatively easy, see #12.

JustinOng, 2022-09-23:
Thanks for getting back! Does look good, I'll attempt to integrate it with my module sometime next week.

stinos, 2022-09-27:
>  I'll attempt to integrate it with my module

With regards to that: now that MicroPython's user module support matured, in the same PR as mentioned above I've simplified what needs to be done to link micropython-wrap into existing code so that a couple of macros are now sufficient, instead of manually having to write the module definition.

stinos, 2022-09-29:
If you come across something not listed in the README please let me know, seems like there can be some peculiarities with ESP32's binary layout and perhaps other things.

JustinOng, 2022-10-01:
Got it to build against micropython e4d90be and ESP-IDF v4.4.2-378-g9269a536ac (From the V4.4 Docker image).

Couple of notes:
- `inline KwargsMaker Kwargs;`: The use of inline variables means that the IDF has to be configured for `gnu++17`. Configuring for `c++17` results in [this](https://github.com/espressif/esp-idf/issues/5897#issuecomment-697257163) error. This was done with the following option in the [module cmake file](https://github.com/micropython/micropython/blob/master/examples/usercmodule/cppexample/micropython.cmake) targeting `gnu++17`: `target_compile_options(usermod_mymodule INTERFACE $<$<COMPILE_LANGUAGE:CXX>:-std=gnu++17>)`
- Exceptions and RTTI have to be enabled in the ESP32's build config (`CONFIG_COMPILER_CXX_EXCEPTIONS=y` and `CONFIG_COMPILER_CXX_RTTI=y`)

stinos, 2022-10-03:
> The use of inline variables means that the IDF has to be configured for gnu++17

Shouldn't that be `c++17` instead? Anyway: my mistake, replaced it with an inline function doing the same but simpler.

> and RTTI have to be enabled

Do you happen to know why RTTI has to be enabled? unix/windows ports build ok here with no RTTI.

> Defining UPYWRAP_USE_EXCEPTIONS=0 UPYWRAP_THROW_ERROR_CODE=0 turns off exceptions except here 

Right, fixed; UPYWRAP_THROW_ERROR_CODE will use UPYWRAP_USE_EXCEPTIONS's value, and exception is replaced with a MicroPython one as should have been the case already.

JustinOng, 2022-10-03:
> Shouldn't that be c++17 instead? Anyway: my mistake, replaced it with an inline function doing the same but simpler.

The warning mentioned using either `c++17` or `gnu++17`, but `c++17` results in a compilation error discussed here https://github.com/espressif/esp-idf/issues/5897#issuecomment-697257163. From that discussion I guessed that the ESP-IDF team might be testing against `gnu++17` which is why I went there.

> Do you happen to know why RTTI has to be enabled? unix/windows ports build ok here with no RTTI.

Building throws this error:
```In file included from /workspaces/myproject/firmware/modules/myproject/module.cpp:3:
/workspaces/myproject/firmware/modules/myproject/lib/micropython-wrap/classwrapper.h: In static member function 'static void* upywrap::ClassWrapper<T>::AsPyObj(upywrap::ClassWrapper<T>::native_obj_t)':
/workspaces/myproject/firmware/modules/myproject/lib/micropython-wrap/classwrapper.h:318:30: error: cannot use 'typeid' with -fno-rtti
       o->typeId = &typeid( T );
                              ^
/workspaces/myproject/firmware/modules/myproject/lib/micropython-wrap/classwrapper.h: In static member function 'static upywrap::ClassWrapper<T>* upywrap::ClassWrapper<T>::AsNativeObjCheckedImpl(mp_obj_t)':
/workspaces/myproject/firmware/modules/myproject/lib/micropython-wrap/classwrapper.h:355:26: error: cannot use 'typeid' with -fno-rtti
             || typeid( T ) != *native->typeId
                          ^
/workspaces/myproject/firmware/modules/myproject/lib/micropython-wrap/classwrapper.h: In static member function 'static void upywrap::ClassWrapper<T>::CheckTypeIsRegistered()':
/workspaces/myproject/firmware/modules/myproject/lib/micropython-wrap/classwrapper.h:647:72: error: cannot use 'typeid' with -fno-rtti
         RaiseTypeException( (std::string( "Native type " ) + typeid( T ).name() + " has not been registered").data() );
                                                                        ^
/workspaces/myproject/firmware/modules/myproject/lib/micropython-wrap/classwrapper.h: In static member function 'static void upywrap::ClassToPyObj<std::function<_Res(_ArgTypes ...)> >::InitWrapper()':
/workspaces/myproject/firmware/modules/myproject/lib/micropython-wrap/classwrapper.h:1109:45: error: cannot use 'typeid' with -fno-rtti
       static wrapper_t reg( typeid( funct_t ).name(), wrapper_t::ConstructorOptions::RegisterInStaticPyObjectStore );
```

I'm afraid I'm not familiar with these features of C++, though I can share my ESP32 build environment if necessary.

Can confirm that `UPYWRAP_USE_EXCEPTIONS=0` works correctly now.

stinos, 2022-10-03:
> Building throws this error:

Hmm, interesting, sort of. I didn't notice after disabling RTTI for msvc because there 'no RTTI' means literally what it says, so `typeid` still works for static compile-time type information i.e. as used here, which isn't really the same as run-time type information (for dynamic_cast etc). And I didn't notice with gcc simply because I put `-fno-rtti` in the wrong place :(

Anyway: thanks for finding; not all of those are really necessary so I made them conditional. If you can't/don't want to enable RTTI, build with 'UPYWRAP_FULLTYPECHECK=0`.

> I can share my ESP32 build environment if necessary

Is that different from the official MicroPython instructions?

stinos, 2022-11-21:
Tested this for a while myself, found no issues, so now in main branch.